spaniakos/AES #1: Compile error with Arduino 1.6.4

DrakeClark, 2016-02-10:
Thanks for a great 8 bit AES library!

I'm getting two simple errors on AES.cpp:
```
In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28:0,
                 from C:\Program Files (x86)\Arduino\libraries\AES-master\AES_config.h:19,
                 from C:\Program Files (x86)\Arduino\libraries\AES-master\AES.h:4,
                 from C:\Program Files (x86)\Arduino\libraries\AES-master\AES.cpp:1:
C:\Program Files (x86)\Arduino\libraries\AES-master\AES.cpp:68:27: error: variable 's_fwd' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 static byte s_fwd [0x100] PROGMEM =
                           ^
C:\Program Files (x86)\Arduino\libraries\AES-master\AES.cpp:88:27: error: variable 's_inv' must be const in order to be put into read-only section by means of '__attribute__((progmem))'
 static byte s_inv [0x100] PROGMEM =
                           ^
Error compiling.
```

This error can be corrected by declaring s_fwd and s_inv to as "const".  Thanks.

spaniakos, 2016-02-11:
Thank you for your notes
I will include the submission in the near future.

closed 2016-09-02 by spaniakos

tscha70, 2018-07-22:
static ...

is now ...

const byte s_fwd [0x100] PROGMEM =
{

spaniakos, 2018-07-23:
seems fine 
does it compile ?