spaniakos/AES #2: missing file avr/pgmspace.h

Urs-Eppenberger, 2016-08-04:
I used your library since it there is a very good tutorial on how to use it on ESP8266 based sensors and Node-Red on Raspberry Pi. I have something usable running already offering authenticity and integrity, but I need your stuff to guarantee privacy too for the communication between my sensors and the home automation central.

I started to compile a sketch using the arduino IDE 1.6.9 on Mac OS X 10.10.5
`In file included from /Users/eppen/Arduino/libraries/AES/AES.h:4:0,
                 from /Users/eppen/Work_Private/Heimautomatisation/Arduino Stuff/TempSensorSec_v0.1/TempSensorSec_v0.1.ino:42:
/Users/eppen/Arduino/libraries/AES/AES_config.h:33:27: fatal error: avr/pgmspace.h: No such file or directory
  #include <avr/pgmspace.h>`
Obviously there is a file missing in my Arduino IDE. It most probably has been included in earlier versions. Do you have an idea on how to solve this?

spicajames, 2016-08-12:
Fixed by replacing the line to #include <pgmspace.h> but new error arise

`.pioenvs/esp01/AES-master/AES.cpp:88:13: error: s_inv causes a section type conflict with __c
￼
static byte s_inv [0x100] PROGMEM =
^
In file included from .pioenvs/esp01/FrameworkArduino/Arduino.h:241:0,
from .pioenvs/esp01/AES-master/AES_config.h:19,
from .pioenvs/esp01/AES-master/AES.h:4,
from .pioenvs/esp01/AES-master/AES.cpp:1:
.pioenvs/esp01/FrameworkArduino/pgmspace.h:21:51: note: '__c' was declared here
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
.pioenvs/esp01/AES-master/AES.cpp:546:12: note: in expansion of macro 'PSTR'
printf_P(PSTR("\n"));
^`

Urs-Eppenberger, 2016-08-24:
In the file AES_config.h I changed
`#include <avr/pgmspace.h>`
to
`#include <pgmspace.h>`

This solved the compile error for my case.

I suppose this has to do with the fact that I need to compile the code for the 'Generic ESP8266 Module' and not with Arduino IDE versions.

Urs-Eppenberger, 2016-08-26:
I had problems with another library, so I was not able to follow up here.
Indeed, I am now stuck with the same compile error as mentioned by spicajames above.
@spicajames:
You offered the solution with two lines of code.
Please help me out here: where exactly do I need to place this code?

spaniakos, 2016-08-26:
I am on vacations . but i will be back stating next week. If you see the
config file there are specific machine compile lines by using if defined
etc. In order to solve and add support for your model i need the board code
( something like the others on the code ). If you can provide the code ad i
will guide you throughout the process.

Thanks spaniakos

spicajames, 2016-08-26:
@Urs-Eppenberger 
The 2 lines are in AES.cpp
`68 static const byte s_fwd [0x100] PROGMEM =`
And 
`88 static const byte s_inv [0x100] PROGMEM =`

Urs-Eppenberger, 2016-08-26:
@spaniakos
your request about the board code is way over my head, I'm very sorry to say. I hope someone else on the forum can be of assistance here. Maybe over at the esp8266.com forum?
@spicajames 
I have seen these two lines. They are not missing. Therefore, where do I need to add what?

spicajames, 2016-08-26:
The original line is missing the "const" word
`before  -->  68 static byte s_fwd [0x100] PROGMEM =`
`after    --->  68 static const byte s_fwd [0x100] PROGMEM =`

Urs-Eppenberger, 2016-08-26:
aahh, yesss.
Now it works. Great.
As the saying goes in Switzerland: wer lesen kann ist deutlich im Vorteil
'those who can read have a clear advantage'

There are now new compile errors, but I have to work on this to exclude my own errors.
Thanks for all the support here.

spaniakos, 2016-08-26:
If there is anything else. I will be able to assist starting Monday or
Tuesday.

closed 2016-09-02 by spaniakos