spaniakos/AES #4: Encryption using the Java gave different result

robertgregor, 2016-09-27:
Hello, I am playing with your library and planning to use it in my IoT esp8266 communication with the server using the java. So on one side, Java is used and on other side, esp8266 chip is used. I have made some small changes, load it to the esp and created the sketch:

#include <AES.h> 
#include <libb64/cencode.h> 

AES aes;
byte *key = (unsigned char*)"0123456789010123";

byte plain[] = "AddNodeAddNodeAd";

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 36753562;
 
void setup() {
  // put your setup code here, to run once:
  Serial.begin (115200);
  delay(500);
  Serial.println("test");
}
void printArray(byte* data, int length) {
       for (int i=0; i<length; i++) {
         Serial.print((char)data[i]);
       }
  Serial.println();
}

void printBase64Array(byte* data, int inputLen) {
       char encoded[base64_encode_expected_len(inputLen)];
       base64_encode_chars((char*)data, inputLen,encoded);
       for (int i=0; i<sizeof(encoded); i++) {
         Serial.print((char)encoded[i]);
       }
       Serial.println();
}

void prekey (int bits)
{
  aes.iv_inc();
  byte iv [N_BLOCK] ;
  byte plain_p[48];
  byte cipher [48] ;
  byte check [48] ;
  unsigned long ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_encrypt(plain,41,cipher,key,bits,iv);
  Serial.print("Encryption took: ");
  Serial.println(micros() - ms);
  ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_decrypt(cipher,48,check,key,bits,iv);
  Serial.print("Decryption took: ");
  Serial.println(micros() - ms);
  Serial.println("\n\nPLAIN :");
  printArray(plain, sizeof(plain));
  Serial.println("\nCIPHER:");
  printBase64Array(cipher, sizeof(cipher));
  Serial.println("\nCHECK :");
  //printArray(check,(bool)true);
  Serial.println("\nIV    :");
  //printArray(iv,16);
  Serial.println("\n============================================================\n");
}

void prekey_test ()
{
  prekey (128) ;
} 
void loop() {
  prekey_test ();
  delay(2000); 
}

Then I wrote the corresponding java code and I expect to get the same result:

public class AES {
  
    public static Key setKey(String myKey) throws NoSuchAlgorithmException, UnsupportedEncodingException {
        MessageDigest sha = null;
        byte[] key = myKey.getBytes("UTF-8");
        key = Arrays.copyOf(key, 16); // use only first 128 bit
        return new SecretKeySpec(key, "AES");
    }
    public static String encrypt(String strToEncrypt, String key) throws NoSuchAlgorithmException, UnsupportedEncodingException, NoSuchPaddingException, 
                                                                            InvalidKeyException, IllegalBlockSizeException, BadPaddingException, InvalidAlgorithmParameterException {
        IvParameterSpec iv = new IvParameterSpec("3675356236753562".getBytes("UTF-8"));
        Cipher cipher = Cipher.getInstance("AES/CBC/NoPadding");
        cipher.init(Cipher.ENCRYPT_MODE, setKey(key), iv);
        
        return Base64.getEncoder().encodeToString(cipher.doFinal(strToEncrypt.getBytes("UTF-8")));
    }

    public static void main(String args[]) throws Exception {
                final String strToEncrypt = "AddNodeAddNodeAd";
                final String key = "0123456789010123";
                System.out.println(AES.encrypt(strToEncrypt, key));
    }    
}

But the results are different. Because with your library I get (base64 encoded):
TfYfvi8/abVK7Ni5x4FpAD82mpq7HY2NAfXHIKX3FTCFMBjdkN4afwUDbDvjYuI0

and with Java I've got:
japwuqYfWgah2eNoWZtbQQ==

I think, the java output looks OK, thus I am encrypting only 128 bits (16 bytes) so should have only 16 bytes output. Why your library is getting 32? Are you padding the input with something?

Thanks.

spaniakos, 2016-09-27:
PKCS7Padding but limited to 16bytes block.
in the library check the AES.cpp for padPlaintext.
the size of the output is resonable, as in the aes.do_aes_encrypt you set
the size to 41.
i know the my implementation is not the easyest to use ( as java ) but its
for micro-controllers with out high level-lang support as java.
for the correct size you can use
i have a framework under way that will make the implementation a lot easier
for many encryption routines, but my job is holding back the completion.

thank you for mentioning that i dont have the padding in the guide. i will
add it in the near future.

keep in mind that the micro-controller con only handle 16bytes block , so
the PKCS7 theoritically can pad from 0 to 255, but for the micro-controller
more than 16 is a waste of memory.

robertgregor, 2016-09-28:
I see, OK, so the padding is clear. However, the first block has to be encrypted anyway in the same way, if the IV is the same. So why I am getting completely different results in java and with your library. I don't understand that, it should be the same...

robertgregor, 2016-09-28:
I have spent some time to look more closely and to match the result of your library with Java. I have found out, that the IV has to be calculated. However, I have modified my sketch now like that:

#include <AES.h> 
#include <libb64/cencode.h> 

AES aes;
byte key[] = "0123456789010123";

byte plain[] = "AddNodeA";
//byte plain[0];

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 36753562;
 
void setup() {
  // put your setup code here, to run once:
  Serial.begin (115200);
  delay(500);
  Serial.println("test");
}
void printArray(byte* data, int length) {
       for (int i=0; i<length-1; i++) {
         Serial.print((char)data[i]);
       }
  Serial.println();
}
void printHexArray(byte* data, int length) {
       for (int i=0; i<length; i++) {
         Serial.print((char)data[i], HEX);
       }
  Serial.println();
}
void printBase64Array(byte* data, int inputLen) {
       char encoded[base64_encode_expected_len(inputLen)];
       base64_encode_chars((char*)data, inputLen,encoded);
       for (int i=0; i<sizeof(encoded); i++) {
         Serial.print((char)encoded[i]);
       }
       Serial.println();
}

void prekey (int bits)
{
  aes.set_IV(my_iv);
  byte plain_p[16];
  byte cipher [16] ;
  byte check [16] ;
  byte iv [16] ;
  aes.get_IV(iv);
  unsigned long ms = micros ();
  aes.do_aes_encrypt(plain,16,cipher,key,bits,iv);
  Serial.print("Encryption took: ");
  Serial.println(micros() - ms);
  Serial.print("PLAIN :");
  printBase64Array(plain, sizeof(plain)-1);
  Serial.print("PLAIN HEX:");
  printHexArray(plain, sizeof(plain)-1);  
  Serial.print("IV    :");
  printBase64Array(iv,sizeof(iv));
  Serial.print("IV HEX:");
  printHexArray(iv, sizeof(iv)); 
  Serial.print("Key   :");
  printBase64Array(key,sizeof(key)-1);
  Serial.print("Key HEX:");
  printHexArray(key, sizeof(key)-1); 
  Serial.print("CIPHER:");
  printBase64Array(cipher, sizeof(cipher));
  Serial.print("CIPHER HEX:");
  printHexArray(cipher, sizeof(cipher));
  Serial.println("============================================================");
}

void loop() {
  prekey (128);
  delay(2000); 
}

And this is the output, I have got:

Encryption took: 414
PLAIN :QWRkTm9kZUE=
PLAIN HEX:4164644E6F646541
IV    :Op6+hkynsr0U7XdOm+dQbQ==
IV HEX:3A9EBE864CA7B2BD14ED774E9BE7506D
Key   :MDEyMzQ1Njc4OTAxMDEyMw==
Key HEX:30313233343536373839303130313233
CIPHER:Op6+hkynsr0U7XdOm+dQbQ==
CIPHER HEX:3A9EBE864CA7B2BD14ED774E9BE7506D

And I can see, that the IV and CIPHER is the same. Why is that? Am I doing something wrong here?

Robert

spaniakos, 2017-01-18:
no, the IV is part of the cipher (this comes from the Mechanism)
the last cipher is the new IV. the data is just confusing, i have to fix that to avoid new confusions.

closed 2017-03-02 by spaniakos

rsegecin, 2017-10-31:
@robertgregor did you solve your issue? I'm having the same problem. I'm ciphering in nodejs and java and they are giving the same result other than on Arduino that I'm having the same result with @spaniakos library or https://github.com/DavyLandman/AESLib library as well.  **padPlaintext** besides "zero padding" to fill the rest of the block with zeros it also considers that you've entered a char array to encrypt that should end with 0x00 (end of string) and the plain, cipher and check buffers should be greater than the NBLOCK by one. But even overcoming this problem I'm still getting different results. Below I'm crypting  in a different manner to avoid the end of string problem.

(nodejs = java) != (@spaniakis = @DavyLandman) 

	#define CBLOCK (1 * N_BLOCK)

	void crypt() {
		AES aes;
		uint8_t key[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15 };
		uint8_t iv[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15 };
		byte tmp_iv[N_BLOCK];

		byte data[CBLOCK];
		byte ciphed[CBLOCK];
		byte deciphed[CBLOCK];

		char auxBuffer[129];

		Serial.println("Encrypting");

		memset(data, 0x00, CBLOCK);
		memcpy(data, "rinaldi", 7);

		memcpy(tmp_iv, iv, 16);
		aes.set_key(key, 128);
		byte test = aes.cbc_encrypt(data, ciphed, CBLOCK / N_BLOCK, tmp_iv);
		
		if (test == SUCCESS)
		{
			Serial.println("Success");

			ByteToHexString(auxBuffer, (unsigned char *)ciphed, sizeof(ciphed));
			auxBuffer[32] = 0x00;
			sprintf(lBuffer, "encrypted-cbc: %s CBLOCK: %i", auxBuffer, CBLOCK);
			Serial.println(lBuffer);
							
			ByteToHexString(auxBuffer, (unsigned char *)iv, sizeof(ciphed));
			auxBuffer[32] = 0x00;
			sprintf(lBuffer, "iv: %s", auxBuffer);
			Serial.println(lBuffer);

			memcpy(tmp_iv, iv, 16);
			memset(deciphed, 0x00, 16);
			test = aes.cbc_decrypt(ciphed, deciphed, CBLOCK / N_BLOCK, tmp_iv);

			if (test == SUCCESS)
			{
				sprintf(lBuffer, "dencrypted-cbc: %s", deciphed);
				Serial.println(lBuffer);
			}
			else
			{
				Serial.println("Failure");
			}
		}
		else
		{
			Serial.println("Failure");
		}

		// Should give
		// 583460271129deead261e621b676f1bf6
		// But it's  giving 
		// 2FD27523B3E1B5A122F397A5F976A680
	}

spaniakos, 2017-10-31:
Can you send me you ino file? Or just the plaintext/iv you want to use? The
expected results are above as i read.

rsegecin, 2017-10-31:
@spaniakos Thank you for the quick reply. I've altered the previous post with the all code and added some nodejs code to compare the output.

spaniakos, 2017-10-31:
Thank you
Will see it tomorrow.

rsegecin, 2017-10-31:
Here it's the same variable values to produce the same output but using the method **do_aes_encrypt**, notice that I've to add 1 to **CBLOCK**.

	#include "AES.h"

	const PROGMEM char HEX_VALUES[] = { '0', '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E', 'F' };

	#define CBLOCK (1 * N_BLOCK) + 1

	char serialBuffer[120];

	void setup() {
		Serial.begin(115200);
	}

	void loop() {
		crypt();
		delay(10000);
	}

	void ByteToHexString(char * hexStrParam, unsigned char * byteArrayParam, unsigned int byteArrayLength)
	{
		unsigned char num;

		for (int i = 0, u = 0; i < byteArrayLength; i++, u++)
		{
			num = byteArrayParam[i] >> 4;
			hexStrParam[u] = (char)pgm_read_byte(HEX_VALUES + num);
			num = byteArrayParam[i] & 0xf;
			hexStrParam[++u] = (char)pgm_read_byte(HEX_VALUES + num);
		}
	}

	void crypt() {
		AES aes;
		byte key[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15 };
		byte iv[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15 };
		byte tmp_iv[N_BLOCK];

		byte data[CBLOCK];
		byte crypted[CBLOCK];
		byte decrypted[CBLOCK];

		char auxBuffer[129];
			
		memset(data, 0x00, CBLOCK);
		memcpy(data, (uint8_t*)"rinaldi", 7);
		
		memcpy(tmp_iv, iv, 16);			
		aes.do_aes_encrypt(data, CBLOCK, crypted, key, 128, tmp_iv);
		
		memcpy(tmp_iv, iv, 16);
		aes.do_aes_decrypt(crypted, CBLOCK, decrypted, key, 128, tmp_iv);

		memset(auxBuffer, 0x00, 129);
		ByteToHexString(auxBuffer, (unsigned char *)crypted, sizeof(crypted));
		sprintf(serialBuffer, "crypted: %s", auxBuffer);
		Serial.println(serialBuffer);

		memset(auxBuffer, 0x00, 129);
		memcpy(auxBuffer, decrypted, 129);
		sprintf(serialBuffer, "decrypted: %s", auxBuffer);
		Serial.println(serialBuffer);

		memset(auxBuffer, 0x00, 129);
		ByteToHexString(auxBuffer, (unsigned char *)tmp_iv, sizeof(tmp_iv));
		sprintf(serialBuffer, "iv: %s", auxBuffer);
		Serial.println(serialBuffer);
	}

rsegecin, 2017-11-01:
I've just spent the all night trying to figure it out why it wasn't encrypting with the same output and I found out that the encrypted data it's not padded PKCS#5. With the string that I passed in the example code I was able to obtain the right encryption padding manually. Updating the line in the example code as shown below:

	memset(data, 0x09, CBLOCK); 
	memcpy(data, (uint8_t*)"rinaldi", 7);

As the block has 16 bytes and the string "rinaldi" has 7 so you've to pad 9 bytes with the byte 0x09 to the right in this block. I've altered the library to automate this process but then I notice that may be I should be padding with 8 bytes instead, I'll be trying this tomorrow.

spaniakos, 2017-11-01:
I will try to run the code in my framework
The weird thing is that i tested the library against standard and it works.

I dont know what time i will have the tests done as in my job it's a hell
week ( server migration and framework update ) haha

deepakchhapru, 2018-12-28:
Hi @rsegecin,

Did you try this, can you please share the code to automate padding. 

Also want to know how can we handle long strings to encypt and decrypt.
I have tried one example which is not working properly. Attaching my code and output
Output is throwing some junk characters too, will be glad if i 
[AES_EcryptDecrypt.txt](https://github.com/spaniakos/AES/files/2714512/AES_EcryptDecrypt.txt)

can get your help on that.

Also I when I try to decrypt the *crypted string* printed in serial monitor from online it is not working, but when I try to decrypt the *iv string* printed in serial monitor i get the results as expected.

> I've just spent the all night trying to figure it out why it wasn't encrypting the same way and I found out that the encrypted data it's not padded PKCS#5. With the string that I passed in the example code I was able to obtain the right encryption padding manually. Updating the line in the example code as shown below:
> 
> ```
> memset(data, 0x09, CBLOCK); 
> memcpy(data, (uint8_t*)"rinaldi", 7);
> ```
> As the block has 16 bytes and the string "rinaldi" has 7 so you've to pad 9 bytes with the byte 0x09 to the right in this block. I've altered the library to automate this process but then I notice that may be I should be padding with 8 bytes instead, I'll be trying this tomorrow.

deepakchhapru, 2019-01-14:
> Hi @rsegecin,
> 
> Can you please share the code to automate padding.
> 
> Also want to know how can we handle long strings to encypt and decrypt.
> I have tried one example which is not working properly. Attaching my code and output
> Output is throwing some junk characters too, will be glad if i can get your help on that.
> 
> Also when I try to decrypt the **crypted string** printed in serial monitor from [online](https://www.devglan.com/online-tools/aes-encryption-decryption) it is not working, but when I try to decrypt the **iv string** printed in serial monitor i get the results as expected.
> 
> [AES_EcryptDecrypt.txt](https://github.com/spaniakos/AES/files/2714512/AES_EcryptDecrypt.txt)
> 
> > I've just spent the all night trying to figure it out why it wasn't encrypting the same way and I found out that the encrypted data it's not padded PKCS#5. With the string that I passed in the example code I was able to obtain the right encryption padding manually. Updating the line in the example code as shown below:
> > ```
> > memset(data, 0x09, CBLOCK); 
> > memcpy(data, (uint8_t*)"rinaldi", 7);
> > ```
> > As the block has 16 bytes and the string "rinaldi" has 7 so you've to pad 9 bytes with the byte 0x09 to the right in this block. I've altered the library to automate this process but then I notice that may be I should be padding with 8 bytes instead, I'll be trying this tomorrow.

Hi @spaniakos,
Can you revert to my query.

rsegecin, 2019-01-15:
@deepakchhapru the code that I made at the time was working only for a block of 16 bytes and I didn't put effort to understand the reason because I decided it was easier to change to Zero padding the rest of my system. If you follow the way I described it will work for the first block from there I think you can come up with a loop to work with all blocks of data.

spaniakos, 2019-01-15:
@deepakchhapru the padding is PKCS#7
there are already function in the library to remove the padding.

When you transfer an encrypted string for decryption, the rule is:
Everything is known except for the key.
That means that you have to transfer the encrypted string along with an unencrypted IV and decrypted with the secret Key.
On top of that i see that you are using memset and memcpy for your code, a practice that is good for memory and speed, but its not recommended as it's already coded in the function of the library with a similar way and tested to work as excepted.
The code and the padding is tested with numerous block sizes.
The size of the padding is automatically calculated from the hex value, and removed if the appropriate function is used.

@rsegecin what was the problem for the code in your case ?

rsegecin, 2019-01-15:
@spaniakos I didn't know that the default padding for this library was different than those set in Java and NodeJs so I was getting different ciphering results. At the time I was researching it I couldn't find a way in your library or any other to set the padding to match for those languages.

deepakchhapru, 2019-01-16:
@spaniakos,
I did decryption in same way u said in online portal using your code given in example, but i did not give me results, hence i used the code shared here by @robertgregor i did the trick.

But same code is not working when i give long plain text to encrypt and decrypt.

So in @robertgregor code i just want to know how to handle long plain text encryption and decryption.
I am using nodemcu for the code shared.
[AES_EcryptDecrypt.txt](https://github.com/spaniakos/AES/files/2763948/AES_EcryptDecrypt.txt)