spaniakos/AES #10: ESP8266 example

anaisortn, 2017-01-30:
Hi, using your example on ESP8266, the decrypted output is `�4���gT�  ��6΄��Add NodeAdd Node��23z���r������͋D��l��պj�N���0��%��렝N���0��%�` instead of `Add NodeAdd NodeAdd NodeAdd NodeAdd Node`

```c
#include "AES.h"

AES aes;

byte *key = (unsigned char*)"0123456789010123";
byte plain[] = "Add NodeAdd NodeAdd NodeAdd NodeAdd Node";
unsigned long long int my_iv = 36753562;

void encrypt()
{
  aes.iv_inc();

  byte cipher[48];
  byte check[48];
  byte iv[N_BLOCK];

  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_encrypt(plain, 41, cipher, key, 128, iv);
  aes.do_aes_decrypt(cipher, 48, check, key, 128, iv);

  printf("check %s \n", (char *)check);
}

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  encrypt();
  delay(1000);
}
```
What am I missing?

spaniakos, 2017-03-01:
please use the provided functions for printing the outputs.
check the Documentation at :
http://spaniakos.github.io/AES/

benjenq, 2017-03-22:
I have the same issue. It seem like a byte-overflow issue on esp8266. The code AES.cpp:

```javascript
void AES::do_aes_encrypt(byte *plain,int size_p,byte *cipher,byte *key, int bits, byte ivl [N_BLOCK]){
	calc_size_n_pad(size_p);
	byte plain_p[get_size()];
	padPlaintext(plain,plain_p);
	int blocks = get_size() / N_BLOCK;
	set_key (key, bits) ;
	cbc_encrypt (plain_p, cipher, blocks, ivl);
}
...
void AES::padPlaintext(void* in,byte* out)
{
	memcpy(out,in,size);
	for (int i = size-pad; i < size; i++){;
		out[i] = arr_pad[pad - 1];
	}
}
```
Then watch & print the parameter "out" value. In theory it  will be
```
[original_plain_text]+[padding]+[padding]+...
```
But in fact it become
```
[original_plain_text]+[padding]+[padding]+...+���렝N���0��%�
``` 
So finally got another value after decrypt.

I need to modify few code in AES.cpp to fix the issue on esp8266:
```javascript
void AES::do_aes_encrypt...
...
byte plain_p[get_size()+1];
plain_p[get_size()] = 0x00;
...
```
to real cut the plain_p (out) value.

I don't know if other mcu has the same issue or not.

spaniakos, 2017-03-27:
the point here is that you are using 
`printf("check %s \n", (char *)check);`
printf function when printing %s (string) are requiring a terminating character
The library cannot predict if the user will use strings or note. it has to stay to a low level byte encoding. thus, i can`t code string function in the library. 
That`s why i have print function in order to print the code based on the size.
after all, in encryption everything except the key is known. thats why when you transmit a cipher, the IV, the cipher and the size are know to everyone.

Taking this as a fact, there is no need to print using string function.

If you really want to use string functions, declare a string with size = cipher + 1, and use low lev
g functions, declare a string with size = cipher + 1, and use low level copy in order to move the buffer into the string followed by the terminating character \0 (0x00). (required by printf %s in order to stop the printing).

spaniakos, 2017-09-29:
closing due to inactivity

closed 2017-09-29 by spaniakos