spaniakos/AES #17: do_aes_decrypt() returns plain text which is having more size then the actual plain text used to encrypt.

arihantdaga, 2017-07-20:
Step 1 . i did do_aes_encrypt().
Step 2. base64 encode the obtained cipher.

Then
on the obtained cipher i did base64_decode().
then i did do_aes_decrypt() with the base64 decode cipher and iv obtained in previous steps.
the resultint plain text obtained from this is of length 80 while the plain text used before Step 1 was only 76 characters long.

spaniakos, 2017-07-24:
please sent me the code used so i can test it.
you can sent it to my e-mail spaniakos@gmail.com

thank you

arihantdaga, 2017-07-29:
There is one more bug now.. 
Encrypting and decrypting of this message results into this-
{"temphumid":"12,12"} -> encrypt -> decrypt again -> {"temphumid":"12,12"?⸮⸮⸮
It also happens if i have 2 consecutive "}}" in the end of the string

spaniakos, 2017-07-30:
iteresting , what board are you using?

arihantdaga, 2017-07-31:
ESP8266 12E

spaniakos, 2017-07-31:
as expected.
there is a bug with ESP8266.
I bought mine a couple fo days ago, but i am leaving for vacation in some
days.

by the end of august i will fully support ESP8266

arihantdaga, 2017-08-17:
@spaniakos Any Updates on it bro ?

spaniakos, 2017-08-17:
Sunday i finish my vacations. Shortly after i will publish an update.

arihantdaga, 2017-08-26:
Is there a formula to predict max cipher size for a given length of message ?
For example if in the following code i dont have to use buffer of size 1000 instead i can allocate buffer based on the msg size. ? 
 
`void encryptMsg(const char* msg, char data_encoded[],char iv_encoded[]){

    char b64data[200];
    byte cipher[1000];
....
...
}

`

yhua537, 2017-09-10:
AES has the same size of input and output.

arihantdaga, 2017-09-10:
@yhua537 What do u mean by size of input and output.. ?
Input and out to do_aes_encrypt() or do_aes_decrypt() ? 
Can u explain ur situation better ?

spaniakos, 2017-09-11:
I would need a sample code and output in order to comment on that.

Thank you

arihantdaga, 2017-12-04:
Any Update on the this... ? 
The issue still persists...

arihantdaga, 2017-12-04:
I think the issue is somewhat with the code only..
Closing for now..

closed 2017-12-04 by arihantdaga

spaniakos, 2017-12-04:
no updates as i am still traveling for some work...
There have been some long months, but by the end of the week i return to my
base station (Home)!
:D thank you for your patience.