spaniakos/AES #18: Not the same result while repeating the same operation

Raphyyy, 2017-07-26:
Hi, thank you for your work :)

Environment : ESP8266 12E 

I tested your librairy 3 times on the same string, but the results are not the same on each.

Here's my code : 

```
#include <AES.h>
#include <Base64.h>

AES aes;

byte key[] = {0x4C, 0x7E, 0x19, 0x16, 0x28, 0x87, 0xD2, 0xA8, 0xAB, 0xF1, 0x15, 0x88, 0x02, 0xCF, 0x4E, 0x3C};
byte iv[] = {0x46, 0xc5, 0x12, 0x05, 0x3f, 0x1c, 0xc7, 0x1d, 0x2a, 0xa1, 0x27, 0xcd, 0xab, 0x7d, 0xcc, 0x9a};

char* message1 = "pmscnv687,232o,0";
char* message2 = "pmscnv687,232o,0";
char* message3 = "pmscnv687,232o,0";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting...");  

  char encryptMsg1[200];
  encryptAES(message1, encryptMsg1);
  Serial.println(encryptMsg1);

  char encryptMsg2[200];
  encryptAES(message2, encryptMsg2);
  Serial.println(encryptMsg2);

  char encryptMsg3[200];
  encryptAES(message3, encryptMsg3);
  Serial.println(encryptMsg3);
}

void encryptAES(char* message, char* output)
{
  aes.set_key(key, sizeof(key));
  
  byte cipher[300];
  char b64data[300];

  memset(cipher, 0x00, 300);
  memset(b64data, '\0', 300);

  Serial.println("===");
  Serial.printf("%s , %d\n", message, strlen(message));

  //int b64len = base64_encode(b64data, message, strlen(message));
  // Encrypt! With AES128, our key and IV, CBC and pkcs7 padding    
  aes.do_aes_encrypt((byte*)message, strlen(message), cipher, key, 128, iv);
  base64_encode(output, (char*)cipher, aes.get_size());
}
```

Here the output : 

```
===
pmscnv687,232o,0 , 16
rxUZg7tFONY5TjW8Kk6ehw==
===
pmscnv687,232o,0 , 16
hC/xnjNW7EiVgLM7u60bmw==
===
pmscnv687,232o,0 , 16
gm9YHW9WE8XPWRZEoJgTqg==
```

The first result is ok while decrypting, the 2 others are not.

I try to initialize the AES object in my encryptAES() function, I had the same result.

spaniakos, 2017-07-26:
there is a bug with ESP8266.
i just got my ESP8266 therefore i hope in the following days i will fully
support the board.

Thank you
spaniakos

Raphyyy, 2017-07-27:
Thanks a lot for doing that.
Could you please tell me an approximate day when it could be fixed ?
It's for an important project and I came accross all Github but your solution seems to be the only one which permit an AES128 CBC and include a padding to crypt variable length message.

spaniakos, 2017-07-27:
hopefully by the end of the next week

alxferraz, 2017-08-10:
Looking forward the next steps in ESP8266! Thanks for the lib! It's Great!

spaniakos, 2017-08-11:
I am off to summer vacations
But soon i will support the board

spaniakos, 2017-09-29:
actually I think that you might have found a little bug there, still trying to understand it ^^

pbabics, 2018-04-02:
Hello, was there any success with adding support for esp8266 ? I looked at this issue and the problem seems to be that `iv` passed to `do_aes_encrypt` gets modified on every call, it is used internally as a buffer for a block.

Maybe I'm not getting the issue, but i can be solved by passing a copy of `iv`,
something like:

```c
void encryptAES(char* message, char* output)
{
  aes.set_key(key, sizeof(key));
 
  byte temp_iv[sizeof(iv)];
  memcpy(temp_iv, iv, sizeof(iv));
 
  byte cipher[300];
  char b64data[300];

  memset(cipher, 0x00, 300);
  memset(b64data, '\0', 300);

  Serial.println("===");
  Serial.printf("%s , %d\n", message, strlen(message));

  //int b64len = base64_encode(b64data, message, strlen(message));
  // Encrypt! With AES128, our key and IV, CBC and pkcs7 padding    
  aes.do_aes_encrypt((byte*)message, strlen(message), cipher, key, 128, temp_iv);
  rbase64_encode(output, (char*)cipher, aes.get_size());
}
```

HitChand, 2018-08-16:
Hello, 
Good Day !!!
Has a solution been found for the issue ?
Even I am not getting a proper output?
Are there any criteria for the input text ?
Following is the sample run on Ubuntu:-

===testng mode

PLAIN :abcdefg**hij**1234**567**890ABCDEFGHIJ

�C�I�C�B�>����P|N��$����۪

Plain2:abcdefg**iij**1234**577**890ABCDEFGHIJ

============================================================
Thanks.

spaniakos, 2018-08-16:
Sent me your .ino file
And I will try to run it on my machine

HitChand, 2018-08-16:
@spaniakos 
Hello,
Where can I send you the code files ?
Thanks.

spaniakos, 2018-08-16:
My email is spaniakos@gmail.com
:)

spaniakos, 2018-08-18:
@HitChand 
are you using ESP8266 or NodeMCU?

HitChand, 2018-08-18:
@Spaniakos

Hello,

I plan to integrate this on my NodeMcu.

However, I have tested the code on Ubuntu.

Thanks & Regards,
HitChand

*** This message has been sent using GIONEE F103 Pro ***

spaniakos, 2018-08-20:
i will check it on a NodeMCU, i am expecting mine to come between 21st and 31st of august (got 2 NodeMCU) that i need for a project.
Since NodeMCU have an ESP8266 core, and since ESP8266 core has a known BUG with my printf.h file i need to mode the library and make it ESP8266/NodeMCU compatible.