spaniakos/AES #19: 'printf_P' was not declared in this scope

facetime88, 2017-09-22:
This works perfect in arduino mega. But when I use it in arduino due, there are several 'printf_P' error:

> AES.cpp:532:46: error: 'printf_P' was not declared in this scope

spaniakos, 2017-09-22:
in AES.h there is the header code that defines functions missing in boards.

`#if defined(__ARDUINO_X86__) || (defined (__linux) || defined (linux))
	#undef PROGMEM
	#define PROGMEM __attribute__(( section(".progmem.data") ))
	#define pgm_read_byte(p) (*(p))
	typedef unsigned char byte;
	#define printf_P printf
	#define PSTR(x) (x)
#else
	#include <avr/pgmspace.h>
#endif
`
you have to add your board code to the line
`#if defined(__ARDUINO_X86__) || (defined (__linux) || defined (linux))`
so it is like
`#if defined(__ARDUINO_X86__) || (defined (__linux) || defined (linux)) || defined(YOUR_BOARD_CODE)`
and I think it will be fine.

spaniakos, 2017-09-22:
my bad the file is : AES_config.h

the code you need for due is : defined(__SAM3X8E__) and is an ARM device
but I need to know if the Arduino due is an AVR or non-AVR board 
I assume you try with the above solution and check if everything works as intended or not and tell me.

add this: 
|| ( defined(__arm__)  &&  defined(__SAM3X8E__) )

as Arduino due is an arm device but the device specific code is __SAM3X8E__

facetime88, 2017-09-29:
Hi,

I have tried the 3 combinations:

`|| defined(arm) `
`|| defined(SAM3X8E)`
`|| ( defined(arm) && defined(SAM3X8E) )`

The error still shows up. Is that okay if I commented out each line of "printf_P(PSTR..." in AES.cpp?

spaniakos, 2017-09-29:
you can try , but there will be other errors as well.
link me your AES_config.h to my email : spaniakos@gmail.com
and I will try to make a modified version (blind coding)

facetime88, 2017-09-30:
Hi,

This is [my file](https://ufile.io/7eibe)
Thank you.

FrancMunoz, 2018-08-18:
Hi! 

To make it work in Arduino MKRGSM 1400, and I guess all arduino ARM family I've changed the following:

Note I've only tested in MKGSM.

So, in Base64.h:

```c++
#include "Base64.h"

#if defined(__AVR__) || defined(__arm__)
	#include <avr/pgmspace.h>
#else
	#include <pgmspace.h>
#endif

...
```

And in AES_Config.h (whole file):

```cpp
/* code was modified by george spanos <spaniakos@gmail.com>
 * 16/12/14
 */

#ifndef __AES_CONFIG_H__
#define __AES_CONFIG_H__

//defined(ARDUINO_SAMD_ZERO) || defined(ARDUINO_SAMD_MKR1000) || defined(ARDUINO_SAMD_MKRZERO) || defined(ARDUINO_SAMD_MKRFox1200) || defined(ARDUINO_SAMD_MKRGSM1400) || defined(ARDUINO_SAMD_MKRWAN1300) || defined(ARDUINO_SAMD_MKRWIFI1010)
#if (defined(__linux) || defined(linux)) && !(defined(__ARDUINO_X86__) || defined(__arm__))
  #define AES_LINUX
  #include <stdint.h>
  #include <stdio.h>
  #include <stdlib.h>
  #include <string.h>
  #include <sys/time.h>
  #include <unistd.h>
#else
  #include <Arduino.h>
#endif

#include <stdint.h>
#include <string.h>

//defined(ARDUINO_SAMD_ZERO) || defined(ARDUINO_SAMD_MKR1000) || defined(ARDUINO_SAMD_MKRZERO) || defined(ARDUINO_SAMD_MKRFox1200) || defined(ARDUINO_SAMD_MKRGSM1400) || defined(ARDUINO_SAMD_MKRWAN1300) || defined(ARDUINO_SAMD_MKRWIFI1010)
#if defined(__ARDUINO_X86__) || defined(__arm__) || (defined (__linux) || defined (linux))
  #undef PROGMEM
  #define PROGMEM __attribute__(( section(".progmem.data") ))
  #define pgm_read_byte(p) (*(p))
  typedef unsigned char byte;
  #define printf_P printf
  #define PSTR(x) (x)
#else
  #if (defined(__AVR__))
    #include <avr/pgmspace.h>
  #else
    #include <pgmspace.h>
  #endif
#endif

#define N_ROW                   4
#define N_COL                   4
#define N_BLOCK   (N_ROW * N_COL)
#define N_MAX_ROUNDS           14
#define KEY_SCHEDULE_BYTES ((N_MAX_ROUNDS + 1) * N_BLOCK)
#define SUCCESS (0)
#define FAILURE (-1)

#endif
```

 @spaniakos maybe you want to include it or I could do a PR... just tell.

spaniakos, 2018-08-20:
i would like to check it on my boards as well.
can you do a PR?
also this is the correct way to do it, as it will show you are a contributor as well :)

FrancMunoz, 2018-08-20:
Sure! let me know if there's any issue with `__arm__` define to find out another to fit...

spaniakos, 2018-08-20:
moving to pull request conv.