spaniakos/AES #23: Encrypt on one arduino and decrypt on another arduino

waghekapil, 2018-05-31:
Hello,

Thanks for the wonderful library. :)

With the following code I'm able to do encryption and decryption on Arduino Uno.

```
#include <AES.h>
#include "./printf.h"

AES aes;

//2b7e151628aed2a6abf7158809cf4f3c
byte key[] = {
	0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09, 0xcf, 0x4f, 0x3c
};

//6bc1bee22e409f96e93d7e117393172a
byte plain[] = {
	0x73, 0x61, 0x69, 0x72, 0x61, 0x6d
};

int plainLength = sizeof(plain);  // don't count the trailing /0 of the string !
int padedLength = plainLength + N_BLOCK - plainLength % N_BLOCK;

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 01234567;

void setup()
{
	Serial.begin(115200);
	while (!Serial) {
		yield();
	}
	printf_begin();
	delay(500);
	printf("\n===testing mode\n");
}

void loop()
{
	prekey_test();
	delay(2000);
}

void prekey(int bits)
{
	//aes.iv_inc();
	byte iv[N_BLOCK];
	byte plain_p[16];
	byte cipher[16];
	byte check[16];
	unsigned long ms = micros();
	aes.set_IV(my_iv);
	aes.get_IV(iv);
	aes.do_aes_encrypt(plain, plainLength, cipher, key, bits, iv);
	Serial.print("Encryption took: ");
	Serial.println(micros() - ms);
	ms = micros();
	aes.set_IV(my_iv);
	aes.get_IV(iv);
	aes.do_aes_decrypt(cipher, padedLength, check, key, bits, iv);
	Serial.print("Decryption took: ");
	Serial.println(micros() - ms);
	printf("\n\nPLAIN :");
	aes.printArray(plain, (bool)true);
	printf("\nCIPHER:");
	aes.printArray(cipher, (bool)false);
	printf("\nCHECK :");
	aes.printArray(check, (bool)true);
	printf("\nIV    :");
	aes.printArray(iv, 16);
	printf("\n============================================================\n");
}

void prekey_test()
{
	prekey(128);
}
```

Since last two days I'm trying to do encryption and decryption on separate arduino boards. After encryption I pass the CIPHER and IV (as it is changing) to the other arduino and do the decryption. But, it is not decrypt properly.

Can you please help me in that?

spaniakos, 2018-06-02:
hello there , 

can you post me the cipher, plaintext, IV and a sample key?
also the code of both arduino would be useful.

thank you !

spaniakos, 2018-06-04:
can you sent me the cipher, iv , key, plaintext and the results you are getting.
it would be helpful if you tell me your boards as well. 

thank you

mudirk, 2018-06-04:
Hi, 

i am using one original arduino nano and one elegoo uno r3 (arduino nano clone).

my data:

mudirk, 2018-06-04:
byte *key = (unsigned char*)"0123456789012";
byte plain[] = "Dies iKlar";
unsigned long myIv = 36753562; 
int bits = 128;

mudirk, 2018-06-04:
Sender
[image]

Receiver
[image]

spaniakos, 2018-06-05:
can you include the code segments of the encryptions and decryption ?

waghekapil, 2018-06-05:
@spaniakos thanks for your reply.

I'm using one Arduino Uno and one RobotDyn's WiFi D1 MINI (ESP8266 16Mb).  As mentioned one Arduino Uno encryption and decryption process is working fine.

Now! I managed to encryption on Arduino and decryption on WiFi module.

**Encryption:**

```
#define KP sizeof(plain) + N_BLOCK - sizeof(plain) % N_BLOCK

byte plain[] = {
	0x73, 0x61, 0x69, 0x72, 0x61, 0x6d, 0x62, 0x68, 0x61, 0x69, 0x6c, 0x6f, 0x67, 0x6f, 0x6a, 0x61, 0x69, 0x73, 0x69, 0x79, 0x61, 0x72, 0x61, 0x6d
};

unsigned long long int my_iv = 01234567;

byte iv[N_BLOCK] = {};
//byte plain_p[KP];
byte cipher[KP] = {};
byte check[KP] = {};

int plainLength = sizeof(plain);

byte key[] = {
	0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09, 0xcf, 0x4f, 0x3c
};

void encryptNow(int bits)
{
	//aes.iv_inc();

	aes.set_IV(my_iv);
	aes.get_IV(iv);
	aes.do_aes_encrypt(plain, plainLength, cipher, key, bits, iv);

	//Plain
	Serial.println("======PLAIN=======");
	for (size_t i = 0; i < plainLength; i++)
	{
		Serial.print((char)plain[i]);
	}
	Serial.println();
	Serial.println("=============");

	//CIPHER
	Serial.println("======CIPHER=======");
	for (size_t i = 0; i < KP; i++)
	{
		Serial.print(cipher[i]);
		Serial.print(" ");
	}
	Serial.println();
	Serial.println("=============");
}
```

**Decryption:**

```
byte key[] = {
	0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09, 0xcf, 0x4f, 0x3c
};

unsigned long long int my_iv = 01234567;

byte iv[N_BLOCK];
byte cipher[32]; // Hard coded size for now
byte check[32]; // Hard coded size for now

void decryptNow()
{
byte bb[] = { 98, 101, 53, 134, 40, 222, 51, 122, 43, 141, 85, 200, 147, 179, 175, 34, 204, 148, 40, 231, 133, 165, 137, 179, 243, 120, 227, 246, 67, 188, 170, 134 }; // Hard coded cipher value from encode function

	aes.set_IV(my_iv);
	aes.get_IV(iv);
	aes.do_aes_decrypt(bb, 32, check, key, 128, iv);

	//CHECK
	Serial.println("======CHECK=======");
	/*for (size_t i = 0; i < sizeof(check); i++)
	{
	Serial.print((char)check[i]);
	}*/
	uint8_t i, j;
	uint8_t loops = aes.get_size() / N_BLOCK;
	uint8_t outp = N_BLOCK;
	for (j = 0; j < loops; j += 1) {
		if (true && (j == (loops - 1))) { outp = N_BLOCK - aes.get_pad(); }
		for (i = 0; i < outp; i++)
		{
			Serial.print((char)check[j*N_BLOCK + i]);
		}
	}

	Serial.println();
	Serial.println("=========================");

	//IV
Serial.println("======IV=======");
	for (size_t i = 0; i < 16; i++)
	{
		Serial.print(iv[i], HEX);
		Serial.print(", ");
	}
	Serial.println();

	printf("\n============================================================\n");
}
```

Here I'm still working on decrypted text string. It is taking extra length as I set the array size to 32 and plaintext size is 24.
Also, for now I've not increment the IV (//aes.iv_inc();).

spaniakos, 2018-06-05:
cool , i will check the algorithm tomorrow as i am not in my house until then.
i will get back to you asap.

sofiacher, 2019-06-06:
> Sender
> [image]
> 
> Receiver
> [image]

I have the same problem. Anyone solve it ?

spaniakos, 2019-06-06:
do you set the same IV in the decrypting arduino ?

spaniakos, 2019-06-06:
can you please tell me the following:
Encryption / Decryption boards and chips,
Confirm the usage of the Same Key and IV in the different algorithms.

Thank you

sofiacher, 2019-06-06:
> do you set the same IV (as the encryption ) in the decrypting arduino ?

Yes, its the same IV, the same key too. I am using 2 arduinos UNO, one for encryption and one for decryption with a serial communication.

spaniakos, 2019-06-06:
can you share the code? either here or @ my email 
spaniakos@gmail.com

spaniakos, 2019-06-06:
resolving though emails.

closed 2019-06-10 by spaniakos

haimilo, 2020-06-24:
Hi i also have the same problems, anyone solve it? can you send me via my email: ngotuehai97@gmail.com. Many thanks!

emirrhan, 2021-07-30:
can you share the code? my mail  emirhanergen1@gmail.com    
looks like a very fun project.

spaniakos, 2021-07-30:
@emirrhan  what code?

emirrhan, 2021-07-30:
@spaniakos Encrypt on one arduino and decrypt on another arduino

spaniakos, 2021-07-30:
the code for that depends on your setup.
it's the same code for both,

in the one you encrypt
then transmit with a method over wifi or over nrf24L01+ antennas for example ( a good library for that is from nrf24 https://github.com/nRF24)

then decrypt the message in an other using the same code.

if you see the code you can split it to encrypt in one and decrypt in the other

Keep in mind that you have to transmit the IV as well and set in after you capture the message.

The key has to be hard-coded in both code-bases

in encryption - decryption the key principal is:
Everything is known expect the key and the plaintext.

emirrhan, 2021-07-30:
how so ?
We use the same code for both encryption and decryption, but has this worked? have you tried it?

emirrhan, 2021-07-30:
@spaniakos 
Have you prepared a project or document? I would be very happy if you could help with this.

spaniakos, 2021-07-30:
i have used that several times.

its literally the same code with one line of code changed
in the 1st use the encrypt method and in the second (the receiver) use the decrpytion method.

in both set the same IV

in the 1st along with Encrypt method use the plaintext
and in the 2nd along with the decryption method use the enrypted text. that's it.

Yes , i have tried it several times with various boards.
I have a codebade (not a project) along with nrf24 antennas.
i haven't encapsulated it into a project, as this library is the encryption library.

It has to be combined with a network library, but since there are various libraries for that i can only make a "project" for the libraries that i am using. Thus the nrf24 libraries.

emirrhan, 2021-07-30:
@spaniakos 
I understood thank you. I will try

spaniakos, 2021-07-30:
initially try the code in one to encrypt
get the encrypted output

and create the decryption hardcoded in the second.

when all is set and working, then you can add the network library and expand from there.

emirrhan, 2021-07-30:
@spaniakos Thanks for your support, I will try to contact you if there is something I cannot do.
I'm ordering the parts now.

spaniakos, 2021-07-30:
@emirrhan sure , anytime ,
You might get me in me time off , but i will reply ASAP when i can

VamsiKrishna-26, 2024-03-11:
if possible share your previous code