spaniakos/AES #29: CBC seems not to work ok

csebe, 2018-10-16:
Hi Georgios,

Thank you very much for your work man!

I tried to encrypt / decrypt an array of bytes using your library and all goes super well if the array is less than 16 bytes. If however the array is more than 16, the bytes after 16 are wrongly decrypted (or encrypted?).

Here is the result of running the attached program with a 25 bytes array.:
`
Plain bytes:
1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,
Encrypted bytes:
198,48,196,191,149,151,125,177,63,56,108,201,80,177,142,152,30,198,187,66,28,99,59,156,171,150,190,97,199,42,242,67
Decrypted bytes:
1,2,3,4,5,6,7,8,9,0,1,2,3,4,5,6,254,83,3,232,34,28,225,4,5
`

I have debugged as much as my skills allow me and it seems that the library calculates correctly there are 2 blocks needed for 25 bytes, but somehow it is not working ok beyond 16 bytes. 

I use ESP8266 (WeMOS mini D1), Arduino IDE and Ubuntu, if it matters.

Thank you in advance for any hints.

Bests,
C.
[WeMOS_AES5.txt](https://github.com/spaniakos/AES/files/2483136/WeMOS_AES5.txt)

spaniakos, 2018-10-16:
i think it's an ESP8266 problem (it has to do with an overridden printf function)
for this particular reason i bought an NodeMCU that incorporates an ESP 8266.

can you sent me your sketch in order to debug it later this week?
you can sent it here or @ spaniakos@gmail.com

thank you

spaniakos, 2018-10-16:
Actually as i am reading your code now, i see that your decrypted variable is 25 length.
This is not good, as the decrypted will have the encrypted size (remember it includes the padding as well) and the padding is required to verify that the plain is not tampered. therefore you have to set it to the encrypted size and the remove the padding (there are functions for this inside the library). The hint about this malfunction is the correct decipher of the 8th and 9th byte of the second part of your cipher. i assume that by cutting one block almost out, the CBC is not chaining correctly, leaving 1/4 of it unchiphered (or uXORd).

@csebe  Please, try it and get back to me
post the output of the screen as well.

csebe, 2018-10-16:
Ahhh!!! Yes, it is is working now, thanks a lot!
And thanks for the hint about removing the padding using functions from the library. In this particular case I know exactly how many bytes are sent, but in other project it might be very useful.

Best regards,
C.

closed 2018-10-16 by csebe