spaniakos/AES #32: Flags for Raspberry Pi

PabloGN, 2018-11-22:
**Error in platform selection**
The if statement doesn't set the correct .h file:
`#if  (defined(__linux) || defined(linux)) && !(defined(__ARDUINO_X86__) || defined(__arm__))`
when doing in Raspberry Pi:
```
sudo make install
g++ -Wall -fPIC -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -c AES.cpp
In file included from AES.h:4:0,
                 from AES.cpp:1:
AES_config.h:19:23: fatal error: Arduino.h: No such file or directory
   #include <Arduino.h>
                       ^
compilation terminated.
Makefile:40: recipe for target 'AES.o' failed
make: *** [AES.o] Error 1
```
Flags defined in my system:
```
cpp -dM ./dummy.hxx |\grep -i -e linux -e __arm__ -e __ARDUINO_X86__
#define __linux 1
#define __linux__ 1
#define __gnu_linux__ 1
#define linux 1
#define __arm__ 1

```
So it compiles when:
`#if  (defined(__linux) || defined(linux)) && !defined(__ARDUINO_X86__) || defined(__arm__)`

**OS**
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
```
Taken from https://blog.hypriot.com/

spaniakos, 2018-11-22:
this is a bug and thank you for mentioning it! i cannot believe that this has slipped form the tests before merging!! 

The correct code should be
`
#if (defined(__linux) || defined(linux) || defined(__arm__) ) && !defined(__ARDUINO_X86__)`

but i would need some confirmation with ARM Arduino devices that i do not have in order to optimize it.
@FrancMunoz  can you please try the library with the above code and report back compatibility with your devices?
This is a major bug and will be fixed ASAP.

spaniakos, 2018-11-22:
BTW @PabloGN  i see that you have RPI v2, if you own a V3 as well can you please compile it there as well? i have an open issue #30 that i cant solve before i buy a new RPi v3

FrancMunoz, 2018-11-22:
Hi! @spaniakos and @PabloGN, it was my fault. I will try with ARM arduino and suggest a condition to be appended to the one that works with other platforms, are you agree?

spaniakos, 2018-11-22:
sure :)
And i will revise my test methods so this wont happen again :)

PabloGN, 2018-11-22:
> BTW @PabloGN i see that you have RPI v2, if you own a V3 as well can you please compile it there as well? i have an open issue #30 that i cant solve before i buy a new RPi v3

I have a RPi3, I'll test as quick as possible :-)

FrancMunoz, 2018-11-22:
I confirm that the corrected code: 

`#if (defined(__linux) || defined(linux) || defined(__arm__) ) && !defined(__ARDUINO_X86__)`

Works with Arduino MKRGSM1400 AND MKRVIDOR. I've compiled, encrypted and decrypted and worked fine.

spaniakos, 2018-11-22:
good, i will test with my arsenal of devices tonight or tomorrow and push the changes!

jaggarwal96, 2018-12-02:
@PabloGN Were you able to get it working on the Rpi 3?

PabloGN, 2018-12-03:
A colleague of mine compiled and tested in RPi 3, so she only had to change the flags to set the libs. Quite faster than RPi2.
So no core dumps, no freeze, no problem.