spaniakos/AES #33: Wrong results when repeating in Raspberry Pi code.

PabloGN, 2018-11-22:
**Plain2 check**
Execution after changing to 3 iterations `  for (int i=0;i<3;i++){ ` in [examples_Rpi/aes.cpp](https://github.com/spaniakos/AES/blob/master/examples_Rpi/aes.cpp), line 22.
Please pay attention to U character in first iteration (TEST TESU TEST TESU) and garbage in the other Plain2 strings.
```
./aes

PLAIN :TESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
Plain2:TESTTESUTESTTESUTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
============================================================
PLAIN :TESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
Plain2:@�'x�HF��m��      VCTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
============================================================
PLAIN :TESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
Plain2:^�y�.D�{��;�g݀�TESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST
============================================================
```
I am measuring execution times to include cryptography in further work based on this one https://doi.org/10.1007/978-3-030-02852-7_16

Thank you again for creating and maintaining this library!

spaniakos, 2018-11-22:
this is due to my neglet of updating RPi example

```
#include <AES.h>
#include "printf.h"

AES aes;

void prekey_test ();
void prekey (int bits, int blocks);

byte key[] = "01234567899876543210012345678998";

byte plain[] = "TESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTESTTEST";
int plainLength = sizeof(plain)-1;  // don't count the trailing /0 of the string !
int padedLength = plainLength + N_BLOCK - plainLength % N_BLOCK;

//real iv = iv x2 ex: 01234567 = 0123456701234567
unsigned long long int my_iv = 01234567;

int main(int argc, char** argv)
{
  printf("\n===testing mode\n") ;

  for (int i=0;i<1;i++){
    prekey_test () ;
  }
}

void prekey (int bits)
{
aes.iv_inc();
  byte iv [N_BLOCK] ;
  byte plain_p[padedLength];
  byte cipher [padedLength] ;
  byte check [padedLength] ;
  unsigned long ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_encrypt(plain,plainLength,cipher,key,bits,iv);
  Serial.print("Encryption took: ");
  Serial.println(micros() - ms);
  ms = micros ();
  aes.set_IV(my_iv);
  aes.get_IV(iv);
  aes.do_aes_decrypt(cipher,padedLength,check,key,bits,iv);
  Serial.print("Decryption took: ");
  Serial.println(micros() - ms);
  printf("\n\nPLAIN :");
  aes.printArray(plain,(bool)true);
  printf("\nCIPHER:");
  aes.printArray(cipher,(bool)false);
  printf("\nCHECK :");
  aes.printArray(check,(bool)true);
  printf("\nIV    :");
  aes.printArray(iv,16);
printf("\n============================================================\n");
}

void prekey_test ()
{
  prekey (128) ;
}
```
try the above example and if it works, please report back (try as many iterators you like)

spaniakos, 2018-11-22:
i have some execution times dated 2 years ago 
[paper_new.pdf](https://github.com/spaniakos/AES/files/2608891/paper_new.pdf)

PabloGN, 2018-11-22:
Ah, Ok, it uses the same lib calls than the arduino code. 
Perfect, it works now on the RPi (all but the Serial part :) )
Thank you for the paper.

spaniakos, 2018-11-22:
i will check the new code as well and push the changes to the example! 
feel free to use the paper (is mine from my thesis)

it used the framework #26 which combines nRF24L01+ antennas in mesh network and the crypto framework, but the code is outdated and not yet published @github

spaniakos, 2018-11-22:
issue will be closed after example code revision and change.